jamct/DoorsignEPD #1: Check gd enabled

noxorius, 2018-01-15:
Hi 
if gd ein NOT enabled the is only a 500 (not responding) return code.
It wold be fine if gd is checked in the server index.php
eg:

@@ -47,6 +47,10 @@
 ^M
 $displayWidth = explode("x",DISPLAYS[$displayType]['size'])[0];^M
 $displayHeight = explode("x",DISPLAYS[$displayType]['size'])[1];^M
+#check GD support
+if ( ! extension_loaded('gd')) {
+       echo "<br>GD support is NOT loaded ";
+}
 $im = imagecreate($displayWidth, $displayHeight);^M
 $background_color = ImageColorAllocate ($im, 255, 255, 255);^M
 $black = imagecolorallocate($im, 0, 0, 0);^M

Best NoXorius

jamct, 2018-01-21:
Good idea. I will add in next update.

jamct, 2018-01-22:
Merged.

closed 2018-01-22 by jamct