jamct/DoorsignEPD #2: Display shows "Wifi not configured!". But Wifi is configured

canoxnet, 2018-01-17:
Hello,

I have now rebuilt the door sign. However, the display shows all the time "Wifi not configured! Connect to hotspot 'ESP32' and open 192.168.4.1" although I have connected to the hostpot, set up the wifi and set the server address (in my case both times http: // 192.168.178.26/server/?debug=true&display=7.5&content=door_sign&scale=28).

jamct, 2018-01-21:
Hello,
which version of basecamp do you use? Please update to the current version.
You have to set "Server to load image from:" (IP or host name) and "Address to load image from:" (with "debug=false". E.g. `"/server/index.php?debug=false&display=7.5&content=door_sign&scale=28"`).

JaquelineP, 2018-02-10:
Hi,

I had the same problem. Are you using the correct wifi library from the ESP? I got a message during compilation that there are multiple wifi.h libraries and arduino would use the default one. After I used the right wifi library the issue was solved.

raz0rsdge, 2018-02-16:
IF the problem still exists, please report the -verbose utput of the Serial console...

closed 2018-03-12 by jamct