jamct/DoorsignEPD #3: Display slightly moved to the right

Tjacksn, 2018-02-02:
Hi, 
I'm trying to build this project and everything is working fine except that the display appears to have an left padding of approx 80 Pixels. So if I choose the Weather Image and 28 scale which should fill out the screen completely, it's cut of on the right. The fun part is, that the cut of part appears in the padding space on the left.
As server I am running Apache 2.4.25 on Debian.

a little help would be greatly appreciated.

best

jamct, 2018-02-02:
Hi,
today some other users had the same problem and we could find a solution (not in this repo yet). Some servers deliver the image in chunks (with the length of the chunk at the beginning). Just tell the ESP32 (line 136 in doorsignEPD.ino) to use HTTP1.0 instead of HTTP1.1

Tjacksn, 2018-02-02:
Works like a charm ... Thank you

jamct, 2018-02-06:
New version adds content-length header to prevent chunking. Could you try HTTP1.1 with your server and new code?

Tjacksn, 2018-02-07:
Yes, it works that way

closed 2018-02-07 by jamct