jamct/DoorsignEPD #5: Display 7.5bwr

danst0, 2018-02-06:
Hi,

Die Einstellung

> #define DISPLAY_TYPE '7.5bwr'

führt bei mir zu einem kontinuierlichen Reboot (jede Sekunde). 7.5 ohne bwr funktioniert.

Viele Grüße,

Daniel

MKdent, 2018-02-06:
Ist bei mir genauso! Leider kann ich nichts mit der dazugehörigen Ausgabe am seriellen Monitor anfangen....

Viele Grüße
Michael

jamct, 2018-02-06:
Hallo,
7.5 bwr ist aktuell ungetestet (noch habe ich kein passendes Display zur Hand). Haben Sie das Gerät mal über USB angeschlossen und auf der seriellen Schnittstelle beobachtet? Steht dort z.B. "Brownout detected"?
Ich werde dem morgen ausführlich nachgehen und melde mich.

danst0, 2018-02-07:
Bei mir ist die Fehlermeldung:
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) flash read err, 1000

MKdent, 2018-02-07:
Hallo, unten sind meine dekodierten Fehlermeldungen zur Ausgabe im seriellen Monitor. Vielleicht wissen Sie was diese bedeuten?

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
abort() was called at PC 0x40084323 on core 1

Backtrace: 0x4008813c:0x3ffe7870 0x4008823b:0x3ffe7890 0x40084323:0x3ffe78b0 0x4008442f:0x3ffe78d0 0x40104582:0x3ffe78f0 0x40102525:0x3ffe7a10 0x400fcd8d:0x3ffe7cd0 0x400f00d0:0x3ffe7d20 0x400f2eee:0x3ffe7d50 0x400828c1:0x3ffe7d80 0x40082909:0x3ffe7da0 0x40007c31:0x3ffe7dc0 0x4000073d:0x3ffe7e30

Rebooting...

0x4008813c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4008823b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40084323: lock_init_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 81
0x4008442f: _lock_init_recursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 95
0x40104582: __sbprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c line 481
0x40102525: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c line 875
0x400fcd8d: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c line 50
0x400f00d0: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 59 (discriminator 8)
0x400f2eee: esp_dport_access_int_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./dport_access.c line 187 (discriminator 1)
0x400828c1: start_cpu1_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 385
0x40082909: call_start_cpu1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 235

jamct, 2018-02-07:
Vielen Dank für den Auszug. Das kann ich bei mir nicht nachstellen. Die 7,5-Zoll-Unterstützung ist noch nicht so lange in der GxEPD-Klasse implementiert. Sind alle Komponenten aktuell?
- Arduino IDE
- Arduino Core für ESP32
- GxEPD und das zugrundeliegende Adafruit_GFX
- Basecamp

MKdent, 2018-02-07:
Außer Basecamp schon. Das GxEPD_SPITestExample Sketch funktioniert....

tne38, 2018-02-07:
Ich hatte die Abstürze mit dem neuen Basecamp. Daher bin ich auf die Version 79db7bd vom 17.1. zurück gegangen.

jamct, 2018-02-07:
Haben Sie die Abstürze auch, wenn das Display nicht angeschlossen ist?

MKdent, 2018-02-07:
Super, Basecamp 0.1.6 geht. Danke für die Hinweise!

Fidel68, 2018-02-26:
Habe seit Samstag auch ein 7.5bwr-Display und das selbe Problem, egal auf welche Basecamp-Version ich in der Arduino-IDE umstelle. Das Test-Sketch funktioniert, wenn ich ohne ‚bwr‘ kompiliere verhält sich das bwr-Display wie das ‚bw‘.
Bei mir soll Display die Daten einer Netatmo Wetterstation anzeigen (bzw. tut das natürlich schon, ich möchte aber gerne ‚kritische‘ Werte in rot anzeigen lassen).

raz0rsdge, 2018-02-27:
Meine Vermutung,da es bei meinem Server nicht läuft, obwohl es korrekt als Bild gerendert wird ist (http://raz0rsedge.de/server/index.php/?debug=true&display=7.5bwr&content=static_image&scale=1), dass die IMAGE-Load Funktion nicht funktioniert. Schließlich laufen die als C-Array eingebetteten Beispielbilder ja auch dreifarbig. Kann also eigentlich nicht am Display-Treiber liegen. Habe auch alle Basecamp-Versionen probiert.

phyreworx, 2018-03-04:
Ich habe ebenso ein 7.5bwr (allerdings mit GELB statt ROT) und laufe auch in diesen Fehler mit dem continuous reboot (ein downgrade auf 79db7bd funzt, allerdings ohne Farbe, das liegt aber vielleicht an dem Gelb?)

Fidel68, 2018-03-04:
So..nachdem ich die 79bd7db nochmal direkt runter geladen habe statt die Bibliotheksverwaltung der Arduino-IDE zu benutzen funktioniert es hier. Interessant ist dabei, das Display zu beobachten. Es kommt nämlich zuerst alles in schwarz und die roten Teile färben sich dann blinkend um.
[image]

raz0rsdge, 2018-03-04:
Yepp das ist normal mit dem Rot-Anteil. Hab bei dem Versuch basecamp 1.8 zu installieren meine komplette IDE zerschossen...nach dem kompilieren steht jetzt bei mir basecamp V 1.8.1.6.1.5 ...nüscht geht mehr

4ndreask, 2018-03-05:
Das scheint wirklich an der aktuellen Version 1.8 zu liegen.
Ich habe das 7.5bwr mit einer alten Version seit über zwei Wochen in Betrieb.
Der Versuch einen neuen ESP32 mit LiPo zu installieren führte zu der bekannten Fehlermeldung.
Nach längerem Probieren habe ich festgestellt, dass der Fehler verschwindet wenn ich auf Basecamp 1.6 zurückgehe.

merlinschumacher, 2018-03-05:
Hallo,

wenn mir jemand mit dem [ESP Exception Decoder ](https://github.com/me-no-dev/EspExceptionDecoder) die Backtrace-Ausgaben des Absturzes schicken kann, werde ich versuchen den Bug in Basecamp zu fixen.

tne38, 2018-03-05:
Hallo,
anbei wie gewünscht die Ausgabe und der daraus dekodierte Backtrace. Die Stelle im Quellcode der ESP-IDF sagt aus, das nicht genügend Heap-Speicher zur Verfügung steht.

entry 0x40078a9c
abort() was called at PC 0x4008285e on core 0

Backtrace: 0x40087f00:0x3ffe3bc0 0x40087fff:0x3ffe3be0 0x4008285e:0x3ffe3c00 0x400829fc:0x3ffe3c30 0x400788ee:0x3ffe3c50 0x40078957:0x3ffe3c80 0x40078a95:0x3ffe3cb0 0x40078bbe:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Decoding 12 results
0x4008285e: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 371
0x40087f00: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x40087fff: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x4008285e: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 371
0x400829fc: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 210

Meine Git-Stände:
espressif-esp32: 70d0d46
basecamp: 9fbb5ff (0.1.8)

phyreworx, 2018-03-06:
@Fidel68 warum hast du GND vom Display an MISO/19 auf dem Lolin angeschlossen? Hab ich irgendwas verpasst?

Fidel68, 2018-03-06:
Das ist eine optische Täuschung. Wenn Du genau schaust sieht es so aus als wären alle Anschlüsse um eins versetzt. Typisches Parallaxen-Problem bei der Aufnahme.

jamct, 2018-03-07:
Das Problem mit der aktuellen Basecamp-Version und dem 7.5bwr ist eingekreist: In Kombination wird zu viel Speicher benötigt (die mehrfarbige Ausgabe braucht ja mehr Bildinformationen), der ESP stürzt ab.
Wir prüfen mal, ob wir in Basecamp noch etwas verschlanken können, um Speicher zu sparen!

phyreworx, 2018-04-22:
Hi, können Sie sagen, ob das aktuelle Basecamp mittlerweile mit dem 7.5bwr funktioniert? Ich habe gerade keine Entwicklungsumgebung zur Hand und es kann es nicht testen. 
> Wir prüfen mal, ob wir in Basecamp noch etwas verschlanken können, um Speicher zu sparen!

toblum, 2018-09-04:
Hallo!

Hat inzwischen jemand eine Kombination gefunden, die zuverlässig mit dem 7.5bwy (oder bwr) funktioniert? Ich hatte weder mit Basecamp 1.8, 1.6 oder "79db7bd" Glück. Immer Dauerreboot. 
@jamct Gibt es denn schon eine Idee, wo abgespeckt werden könnte?

Viele Grüße
Tobias