jamct/DoorsignEPD #7: entferne eraseDisplay

noxorius, 2018-02-09:
Hi
in der doorsignEPD.ino 
Zeile 166
Ich hab mich schon länger gefragt warum das display zuerst gelöscht werden muss und dann upgedatet.
wenn Du das eraseDisplay(true); mit einen delay(3000); ersetzt wird das display upgedatet ohne vorher gelöscht werden zu müssen.

Anscheinen hat der http client eine delay wenn Du ihn zu früh abfragst ist das Ergebnis leer --> kein Bild auf dem Display.

JaquelineP, 2018-02-10:
Hi,

kann ich bestätigen. Funktioniert sehr gut und der Bildaufbau geht schneller.

jamct, 2018-02-10:
Hallo,
werde ich testen und in den Code aufnehmen.
@noxorius Den HTTP-Client werde ich umstellen auf einen anderen Client mit Callbacks und auf das Ergeignis "Daten abgerufen" reagieren. Das ist die sauberere Lösung.

MKdent, 2018-02-11:
Hallo,
bei mir wird,  nach dem Ersetzen von eraseDisplay(true) auf delay(3000), das Display einmal aktualisiert und bleibt dann aber auf weiß. Es wird kein neues Bild aufgebaut. Stimmt es nur den Befehl zu ersetzen?
Was in diesem Zusammenhang vielleicht passt: Wie oben beschrieben hat der http client ein Delay. Bei mir kommt es im Production Modus  vor, dass das Bild einige Male aktualisiert wird, dann aber nicht mehr. Laut Ausgabe im seriellen Monitor, erwacht zwar der esp32, aber es kommen keine Bild-Daten mehr an. Ich bekomme nur die Ausgabe "image loaded". Liegt das an meinem Server/Router. Hat jemand selbiges beobachtet?

commit 23967054 (2018-03-06): Better readability, some more comments, set timeout, rm clear screen (resolves #7)
closed 2018-03-06 by jamct