jamct/DoorsignEPD #11: change sleeptime dynamicly

JaquelineP, 2018-02-15:
Is there a possiblity to change the sleeptime dynamicly like it is done with the Production mode?

`//To stop productionMode (no deep sleep, web config), set http-header X-productionMode: false 
#header("X-productionMode: false");`

If Yes: Please provide the required Header or puiblish in which file/library the Header is processed.

jamct, 2018-02-15:
Not implemented at the moment. Will add it to my todo!

d-gattermann, 2018-02-15:
Similar request: reduced update interval during the night (when nobody is seeing changes) anyway. Bonus points for a different schedule on weekends.

jamct, 2018-02-15:
I think we need a control surface for this ideas. I like it: Manage sleep times and contents for multiple displays. Will think about it.

JaquelineP, 2018-02-17:
I think if there is a header available (like for production-mode) it is pretty straightforward. 

The Header transmits the next sleeptime to the ESP-32 and everything else can be configuered in php.

what is unclear to me is weather the Header information needs to be in the index.php or can also be in the content files (conference-room.php etc.). I will test that shortly. If it is possible to write it in the content files as well, every display can have different sleeptimes configuered in the coresponding content.php file.

It is also easy in php to find out whether it is a workday or not so the request from d-gattermann is easily implemented as well without the need for a control surface.

JaquelineP, 2018-02-17:
It works with aditional header lines in the index.php file and a small adition in the DoorsignEPD.ino file. I will create a pull-request tomorrow if I have time.

In Index.php:
`if(intval(date("H"))>= 6 and intval(date("H"))<=8){
	header("X-sleepTime: 60");
}
elseif(intval(date("H"))> 23 or intval(date("H"))<6){
	header("X-sleepTime: 3600");
}
else{
	header("X-sleepTime: 300");
}
if(date("l")=="Saturday" or date("l")=="Sunday"){
	header("X-sleepTime: 120");
}`

in DoorsignEPD.ino after line 186:

`        if (header.indexOf("X-sleepTime") > 0) {
        iot.configuration.set("ImageWait",header.substring(header.indexOf("X-sleepTime")+13,header.indexOf("X-sleepTime")+15));
        }`

jamct, 2018-02-17:
Nice work!

JaquelineP, 2018-02-17:
@d-gattermann for your weekend/night case you could use something like:

```
if(intval(date("H"))> 23 or intval(date("H"))<6){

	header("X-sleepTime: 3600");

}

else{

	header("X-sleepTime: 300");

}

if(date("l")=="Saturday" or date("l")=="Sunday"){

	header("X-sleepTime: 120");

}
```

d-gattermann, 2018-02-20:
@JaquelineP Thanks!

jamct, 2018-03-15:
Feature is added by #22. Thanks to @jdede

closed 2018-03-15 by jamct