jamct/DoorsignEPD #17: Akku-Anzeige

Fidel68, 2018-03-04:
Ich habe auch ein Lolin32 und einen 2500mAh Flachakku angeschlossen, mit 10 min sleeptime bin ich damit auf gut eine Woche Laufzeit gekommen. Ich überlege mir jetzt, die Akku-Messung aus dem Türsensor-Projekt mit einzubauen und den Wert auf das Display zu bringen. Ich bin mir nur nicht sicher, was dazu der beste Weg ist... im ESP-Code direkt auf‘s Display schreiben oder den Wert z.B. in die URL einbauen und vom index.php in die Grafik zeichnen lassen?

jamct, 2018-03-04:
Ich würde den Wert mit dem Aufruf per GET an den Server übertragen. Dann kann man später die Werte auch loggen oder Warnungen ausgeben, wenn ein Akku zur Neige geht.

borgelite, 2019-05-09:
Könnte mal jemand ein beispiel aufzeigen ? Ist, das im Code schon enthalten ? Wenn ja wie lautet die URL für die GET abfrage ?

jdede, 2019-05-09:
Ein Beispiel ist noch nicht vorhanden bzw. ggf. im Türsensor-Projekt zu finden (https://www.heise.de/select/ct/2018/2/1515447393533211). Jedoch sind noch zwei Punkte zu beachten:

- Im Türsensor-Projekt wird ein Spannungsteiler am Analogeingang verwendet, um den Ladezustand der Batterie zu bestimmen. Bei vielen Batterietypen variiert die Spannung der Zelle nur gering im Vergleich zur verbleibenden Ladung, so dass sich nur schwer von der Spannung auf den Ladezustand schließen lässt. Es handelt sich hierbei eher um eine grobe Schätzung.
- Überträgt man Messwerte an den Server, sollte man nicht die `GET`-Methode verwenden: `GET` ist idempotent, d.h. der Aufruf der gleichen URL darf auf dem Server nichts verändern. Würde man Daten via GET übertragen (bsp. refresh im Browser), würde jeder Aufruf einen neuen Eintrag in der Datenbank erzeugen.

Die korrekte Methoden um Daten auf den Server zu übertragen, sind `PUT` oder `POST`. Dies kennt man auch aus dem Browser: Lädt man eine Seite mit einem Formular neu, fragt dieser nach, ob die Daten wirklich erneut gesendet werden sollen.

Warum das Ganze wichtig ist?
1) Guter Stil
2) Wenn ein Bot (z.B. der GoogleBot) auf eine Seite stößt, folgt dieser Links (also `GET`-Referenzen) -- diese sind ja idempotent. Ändere ich nun aber Daten auf dem Server via `GET`, so kann der Bot dafür sorgen, dass innerhalb kürzester Zeit meine gesamte Datenbank voll geschrieben wird (gerade bei dynamischen Dokumenten).

Zurück zur Frage: Der Aufbau sieht ganz grob wie folgt aus:

- Spannungsteiler aus zwei Widerständen. `V_Bat -- R1 -- R2 -- GND`. Dieser sollte so ausgelegt sein, dass die Spannung zwischen R1 und R2 die maximale Eingangsspannung des Microcontrollers nicht übersteigt. Im Falle des ESP32 wären das 3,3V. Nutzt man nun eine Batterie mit z.B. 4,7V und einen Spannungsteiler mit 2x 10kOhm liegen maximal 2,35V zwischen den beiden Widerständen an. (Eigentlich gehört da noch ein wenig Schutztechnik etc. mit dazu. Aber für den Anfang reicht es.)
- Diese Spannung kann nun über einen analogen Eingang eingelesen werden: Einfach eine Verbindung zwischen den beiden Widerständen und einem analogen Eingang herstellen.
- Dieser Wert kann dann (wie im Türsensor beschrieben) über `analogRead()` ausgelesen und entsprechend verarbeitet werden (-> via `POST` zum Server oder irgendwie anders auf dem ESP verwenden).

Übrigens: Eine so genannte _Fuel Gauge_ liefert deutlich genauere Ergebnisse als ein Spannungsteiler.

Hoffe, dass das hilft.