jamct/DoorsignEPD #18: Pythonversion des Servers

danst0, 2018-03-05:
Hallo,

ich ersetze gerade den Server durch mein eigenes Python Script. Soweit habe ich mit bottle schon einiges hinbekommen. Ich scheitere im Moment allerdings an dieser Funktion:

>  $bytes .= pack('H*', str_pad(base_convert($bits, 2, 16),2, "0", STR_PAD_LEFT))

Hat jemand eine Idee wie man das in Python schreibt?

Viele Grüße,

Daniel

jamct, 2018-03-07:
Ich bin nicht so tief in Python, dass ich fertigen Code liefern könnte. Aber ich kann zumindest erklären, was ich mir bei der Zeile gedacht habe (die ist ja alles andere als selbsterklärend):
- die innere Funktion base_convert wandelt die Ketten aus Bits in Hex-Werte um und str_pad füllt sie mit Nullen auf (sonst würde aus 0A nur A und es würde sich alles verschieben) .
- Pack verwandelt die Hex-Werte in Bytes, die an den Byte-String angehängt werden.

danst0, 2018-03-07:
Danke,
bis jetzt habe ich: 
```
def bitstring_to_bytes(s):
    return int(s, 2).to_bytes((len(s)+7) // 8, byteorder='big')

my_bytes += bitstring_to_bytes(bits)
```
Die Funktion wandelt den Bit-String in einen Integer um, der dann wiederum in ein Python Bytes umgewandelt wird mit ganzzahliger Byte-Länge.
Von meinen Experimenten auf Python/PHP-Fiddle zeigen sich allerdings kleine Unterschiede:
PHP:
```
$bytes = pack('H*', str_pad(base_convert("110001111", 2, 16),2, "xx", STR_PAD_LEFT));
echo bin2hex($bytes);
```

Python:
```
s = '110001111'
tmp = int(s,2).to_bytes((len(s)+7)//8, byteorder='big')
print(tmp)
print(tmp[0])
print(tmp[1])
```

PHP Ergebnis: 18f0
Python Ergebnis: b'\x01\x8f' bzw. 1 und 143

Ich verstehe nicht, wo PHP die Null am Ende nimmt. Es ist tatsächlich nicht die 0 aus dem str_pad, die habe ich ja durch x ersetzt...

Daniel

jdede, 2018-03-09:
Hallo Daniel,
hab mir diesen Teil des Codes nicht im Detail angeschaut. Ich denke aber, dass du ein Problem mit deinem Beispielarray hast: `110001111` ist 9 bit lang und entspricht dezimal 399 und hexadezimal 018f (wie in deinem Python-Snippet zu sehen). Im PHP-Code stellt die Modulo-Operation sicher, dass du nicht mehr als 8 bit konvertierst:

https://github.com/jamct/DoorsignEPD/blob/dfccdf5a7e2602a111b1058e46d470d3943f93bc/server/index.php#L167-L170

Wenn du die Funktion mit 8 bit testest, passt es. Alternativ kannst du die PHP-Version auch für 16 bit hex wappnen:

    str_pad(base_convert("110001111", 2, 16),4, "0", STR_PAD_LEFT))

Dann stimmt das Ergebnis auch wieder.

danst0, 2018-03-13:
Super, jetzt hat es geklappt!

closed 2018-03-13 by danst0