jamct/DoorsignEPD #19: Not showing correct SSID of AP

jamct, 2018-03-07:
- Basecamp now supports variable SSID, depending on MAC-address. When display is not configured, it says: "Connect to hotspot 'ESP32'".
- Display always shows secret (even if no AP-secret is set)

jdede, 2018-03-09:
Is there a way to readout the current SSID in Basecamp? When I checked last (two weeks back), there was no (nice) option to get the current SSID when in AP mode.

Regarding the secret: Is the value completely empty if no secret is set?

I will check both issues as soon as I have access to my board.

jamct, 2018-03-09:
My colleague @merlinschumacher (maintainer of Basecamp) is working on a solution in Basecamp. If you created a solution, you can send him a PR.
The secret is always created but only set if activated in basecamp-constructor.

jdede, 2018-03-10:
Okay, now I got the same problem. Didn't upgrade to Basecamp 0.1.8...
After checking the code again, I created a PR for the missing feature (https://github.com/merlinschumacher/Basecamp/pull/44). Let's see, when/if it gets merged and is available in the Arduino library manager.