jamct/DoorsignEPD #24: ESP32 wacht nicht auf

rocklobster42195, 2018-03-16:
Hallo.
Der ESP wacht nicht mehr auf, ich kann ihn eigentlich nur mit `setupMode=true` laufen lassen. Dann bekomme ich auch das LOG angezeigt. Bei `setupMode=false` erscheint im seriellen Monitor erscheint nur:
`.
.
.
.
.
.
Wait till the client is connected
Going to deep sleep now...
`
Selbst ein Aus- und wieder Einschalten, veranlasst den ESP32 nicht, die Grafik anzufordern.
Ich habe es bereits mit und ohne `X-sleepInterval:60` im Header versucht. 
Danke für die Hilfe.
Viele Grüße
Boris

rocklobster42195, 2018-03-16:
M.E. fehlt da ein `tcpClientConnectionInProgress=true;` beim `client.connect` und die entsprechende Abfrage bevor der ESP32 schlafen gelegt wird.

rocklobster42195, 2018-03-16:
Es ginge auch `while (!tcpClientConnected){}` nach `Serial.println("Wait till the client is connected");`

jdede, 2018-03-17:
Moin, jupp. In Zeile 382 fehlt ein `tcpClientConnectionInProgress=true;`. `while (!tcpClientConnected){}` wäre busy-waiting und würde das async-Konzept zerstören. Hab gerade keine Hardware zum testen hier, werde das aber dann morgen oder am Montag testen und einen PR auf machen.

D3viloper, 2018-03-19:
Guten Morgen, Ich habe das selbe Problem, Ich finde es jedoch sehr komisch, dass ich die Meldung "OnConnect" erst bekomme nachdem ich die Meldung "Going to deep sleep now" bekomme. (Den Code habe ich in Zeile 382 angepasst wie oben erwähnt.)

rocklobster42195, 2018-03-19:
Denk daran, dass du in der Abfrage (Zeile 390) auch noch `|| tcpClientConnectionInProgress` ergänzt, wenn Du `tcpClientConnectionInProgress=true;` eingefügt hast.

D3viloper, 2018-03-19:
Erstmals Danke für deine Hilfe,
Ich habe es noch hinzugefügt (Zeile 390) und es halt so halb funktioniert, jetzt habe ich das Problem das er nicht schlafen geht weil er angeblich nicht im Produktion Mode ist obwohl die Option aktiviert ist/wurde.

rocklobster42195, 2018-03-19:
Probiere doch mal in der `index.php` des Servers die Zeile 3 einzubeziehen ("#" am Anfang löschen). Damit aktivierst du den "production mode" über den header.

D3viloper, 2018-03-19:
Funktioniert jetzt alles, danke für die Hilfe!

jdede, 2018-03-19:
Guten Abend,
so, habe einen PR erstellt. Das Problem tritt bei mir nun nicht mehr auf. Gibt es noch Kommentare oder Anmerkungen?

rocklobster42195, 2018-03-19:
Danke. Bei mir ist alles gut.

D3viloper, 2018-03-20:
Danke, bei mir ist auch alles gut

commit b5a3cf95 (2018-03-20): This PR fixes #24 where the ESP will not download data after deep sleep. It also includes some minor debug improvements.
closed 2018-03-20 by jamct