jamct/DoorsignEPD #27: Ethernet

ger82, 2018-06-08:
Kann man das Projekt auch anpassen so das es mit Ethernet funktioniert?

jdede, 2018-06-08:
Grundsätzlich schon. Jedoch unterstützt der ESP32 nur Bluetooth und WiFi. Ethernet müsste über ein entsprechendes Shield verwirklicht werden. Weiterhin müsste Basecamp angepasst werden. Technisch möglich, jedoch nichts, was man *mal eben* in 20 Minuten schreibt.

Hier ist ein Link für Ethernet auf dem ESP32: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

Für Ethernet würde ich ggf. auf einen RaspberryPi wechseln. Hier hat man dann eine deutlich leistungsfähigere Plattform, die Waveshare Displays werden auch hier unterstützt und mit dem entsprechenden PoE-Hat könnte man das Ganze auch via PoE mit Spannung versorgen.

ger82, 2018-06-08:
Ich habe das OLIMEX ESP32-EVB.
Geht es auch ohne das Basecamp?

jdede, 2018-06-08:
Klar. Dann musst du dich jedoch um die ganzen Komfortfunktionen, die dir Basecamp zur Verfügung stellt selbst kümmern. Die Bibliothek für das Display ist unabhängig und hier verfügbar: https://github.com/ZinggJM/GxEPD

ger82, 2018-06-08:
Ich glaube es hin bekommen zu haben, nur leider funktioniert die Anzeige auf dem Display nicht. Mit dem Board haben Sie nicht zufällig Erfahrung?

jdede, 2018-06-11:
Ich habe keine Erfahrung mit dem Board, würde jedoch die Verkabelung prüfen: Gibt es evtl. Doppelbelegungen mit dem Display / Ethernet? Hier würde ich einen besonderen Blick auf den CS-Pin werfen und ggf. einen anderen Pin testen.

ger82, 2018-06-18:
Auch dieser Tipp war ein guter Wegweiser, ich musste die PIN teilweise umbelegen aber nun funktioniert es.
Für alle die ebenfalls nach einer Lösung suchen: in der Datei ...\Arduino\hardware\espressif\esp32\variants\esp32-evb\pins_arduino.h (Das selbsterstelle Verzeichnis, was zu Beginn des Projektes aufgebaut wird) habe ich folgendes definiert
static const uint8_t SS    = 5;
static const uint8_t MOSI  = 2;
static const uint8_t MISO  = 15;
static const uint8_t SCK   = 14;
Sodass nun diese GPIOs genutzt werden:
Busy - 4
RST - 16
DC - 17
CS (SS) - 5
CLK (SCK) - 14
DIN (MOSI) - 3.

closed 2018-06-18 by ger82