jamct/DoorsignEPD #28: compiling error the code

dodo124, 2018-10-28:


jamct, 2018-11-21:
You ran into a lot of problems with dependencies of this project.
I am currently refactoring this project. My experiments with Platform.io are very promising. Compared to Arduino IDE it has a per-project dependency management!
I will keep you informed if there are any news.

jdede, 2019-01-14:
Which version of ArduinoJson do you have installed? I guess it is version 6.x... This is still beta and not very stable. Furthermore, it has several API changes. Downgrade to the latest 5.x version (should be 5.13.4) via the library manager -- and don't upgrade it if suggested by the board manager... ;-) This should solve your problem.