jamct/DoorsignEPD #30: Error: -55

PaulG4H, 2019-01-14:
Hello,

after successfully compile with Arduino IDE 1.8.8 in Linuxmint and upload to an Olimex ESP32-EBV I just get this serial output:
`Basecamp V.0.1.6
MAC-Address: 24:0a:c4:17:5f:6c, Hardware MAC: 24:0a:c4:17:5f:6c
*******************************************
* ACCESS POINT PASSWORD: NhU/,:yK
*******************************************

Wait till the client is connected
Not going to deep sleep. Reason: Not in production mode
Connection setup in progres
Error:-55
transmitDone
OnDisconnect
transmitDone
Not going to deep sleep. Reason: Not in production mode
Wait till the client is connected
Not going to deep sleep. Reason: Not in production mode
Connection setup in progres
Error:-55
transmitDone
OnDisconnect
transmitDone
`
The setup page looks like this:
[image]

Here are the test of the webserver (Apache 2.4) with debug=false
[image]

and debug=true:
[image]

btw. the display is connected to the pins like shown here:
[image]

But there is nothing shown on the display at all...

Maybe someone can point me to the right direction where to dig...

jdede, 2019-01-14:
There is a problem with the name resolution. Error -55: "DNS Failed" (-> from AsyncTCP). Please check if it works with the IP address instead of the hostname.
The server seems to work fine: With debug=true, you get the png image, with debug=false the format required by the microcontroller.

PaulG4H, 2019-01-14:
Hello jdede,

many thanks for your incredible fast reply!

but even after I figured out how to set the default virtualhost config to answer on the servers ip, the error is still the same.

For me curios is that I do not get an error when I disconnect the e-paper display and until now I never get an update on it still the image shown when I get it...

The two switches on the e-Paper HAT for Raspberry Pi are set as shown on the images I found (B / 0)

Maybe someone are so kind to post an detailed picture how the display is connected?

Many thanks in advance!

jdede, 2019-01-14:
Hmm, could be something with a double-used pin. I just checked the schematic of your board (https://github.com/OLIMEX/ESP32-EVB/raw/master/HARDWARE/REV-D/ESP32-EVB_Rev_D.pdf) and it seems to have quite a lot of additional stuff connected to the ESP32: Ethernet, CAN, Button, IRDA, SD-Card...
In case of my ESP32, I had to connect the chip select pin (cs) to a different one and everything worked out.
Which switch do you mean? On my display, I only have a soldering pad to select between three and four wire SPI. In my case, four-wire-SPI is selected (default) and it works for my two displays (BS:0).

PaulG4H, 2019-01-15:
Hello jdede,

many thanks again, I am an ESP "beginner" and this sounds absolutely logical!
I do have an Olimex ESP32 PoE also here and will check it this soon an will update this post.

jdede, 2019-01-15:
The ESP32-PoE also comes with a lot of peripherals connected to it... But it should be possible with both boards as the ESP32 has plenty of GPIOs and communication ports.
I personally preferred a minimized version with less peripherals for my first steps with the ESP32. I have good experience with the ESP32 Thing by SparkFun: Well documented, good design, good support, good tutorials but with roughly 20€ quite expensive compared to others.
Good luck!

PaulG4H, 2019-01-15:
Hello, 
i gave up with the Olimex ESP32-EVB because I do not find which Gpio's are not used and try with

Olimex ESP32-PoE, but here most of the ports which are needet not available, there are just two connectors with 10 pins and three of them for Power so just 17 GPIO's and I do not find in the ino where to change the gpio's needed as example GPIO 18?

<code>#define CHIP_SELECT 5

GxIO_Class io(SPI, CHIP_SELECT, 17, 16); 

GxEPD_Class display(io, 16, 4);</code>

[image]

jdede, 2019-01-15:
I dived into the codes and checked the definition and mode of the pins. The lines BUSY, RST, DC and CS are normal GPIOs and handled via the `digitalWrite` / `digitalRead` functions. Those can be connected to arbitrary pins (w/o pullup / pulldown resistors etc.). The SPI port is defined by the hardware design of the microcontroller. SPI1 of the ESP32 is on pin 18 and 23 and (somehow) fixed. SPI can handle several devices on one interface. The active device is selected by the CS (chip select) GPIO.

    Name		GPIO		Description
    -----------------------------------------------------------------------
    BUSY		4		Display is busy, GPIO, Input
    RST		16		Reset Display, GPIO, Output
    DC		17		Data / Command selection, GPIO, Output
    CS		5		CS, Chip Select, GPIO, Output
    CLK		18		SPI SCK pin (clock)
    DIN		23		SPI MOSI pin
    GND		GND		Ground
    3V3		3V3		Voltage 3V3

I went through the docs of the ESP32-EVB. The following ports should work out as GPIOs:

- GPIO 6
- GPIO 7
- GPIO 8
- GPIO 9
- GPIO 10
- GPIO 11

They all should be available on the extension header. Maybe try the following wiring:

    Name		GPIO
    ---------------------
    BUSY		6
    RST		7
    DC		8
    CS		9
    CLK		18
    DIN		23
    GND		GND
    3V3		3V3

Hope that helps.

PaulG4H, 2019-01-16:
Hello jdede,

thank's for your detailed help but now I get stucked again with an new behavior!

here are the output of the serial console:
```
Basecamp V.0.1.6
MAC-Address: 24:0a:c4:17:**:**, Hardware MAC: 24:0a:c4:17:**:**
*******************************************
* ACCESS POINT PASSWORD: Nh*****K
*******************************************

Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
Not going to deep sleep. Reason: Not in production mode
In setup mode
e⸮Rչ⸮2⸮L⸮⸮⸮⸮ҵS⸮
⸮⸮.⸮⸮⸮LB=W⸮⸮⸮⸮%⸮⸮Uɒ⸮⸮W⸮⸮⸮b⸮iU⸮⸮⸮*⸮e1⸮⸮⸮%⸮⸮5⸮
⸮+⸮⸮⸮N⸮宱 ⸮⸮⸮C⸮W⸮⸮⸮빣ڷj
eWA⸮⸮⸮⸮0L⸮⸮⸮⸮C⸮H⸮'⸮E⸮BR⸮C⸮⸮⸮
U⸮⸮ѪU⸮)⸮⸮t'⸮⸮b⸮i}F⸮*⸮e1A*⸮%⸮⸮U⸮
⸮+͡⸮.V⸮ Y..	⸮0⸮*⸮YW⸮⸮⸮⸮9c⸮⸮⸮j⸮e⸮.R⸮⸮ &⸮⸮⸮⸮Ҳ⸮⸮⸮
!H⸮.⸮⸮xL⸮⸮*⸮⸮DեQC⸮jTUQe⸮⸮⸮⸮:⸮E⸮BZ⸮I%⸮⸮⸮%⸮A*⸮%⸮⸮5⸮
f+⸮⸮⸮d*ɲ	ʐ⸮⸮!w}⸮aZ⸮)A⸮Sj
e⸮.R⸮n⸮ &⸮⸮⸮⸮⸮⸮ҵ⸮H⸮⸮⸮⸮:⸮E0B⸮Q⸮ה⸮I⸮⸮%UME⸮JI⸮o⸮' ⸮⸮PRe1⸮⸮⸮%=⸮⸮JH⸮+͡ ⸮⸮⸮e.q⸮L⸮⸮C⸮⸮⸮[⸮⸮.⸮⸮7⸮j
⸮w⸮⸮] &ʶ⸮º⸮ҵ⸮H⸮⸮⸮⸮'⸮⸮0⸮IE
⸮⸮⸮%Q⸮ߪ⸮U⸮⸮	⸮⸮⸮x,⸮H*⸮}⸮LP*⸮%=⸮TJ!+⸮h⸮⸮a*⸮⸮,⸮L⸮⸮
⸮Y⸮}m⸮Z⸮⸮⸮3⸮⸮⸮ׁ⸮8⸮⸮⸮⸮⸮⸮ҺMSH⸮⸮⸮⸮i0⸮BRU⸮⸮⸮%Q⸮⸮*⸮UQ⸮,⸮⸮i0⸮BBM⸮Ie⸮⸮*⸮%⸮⸮⸮⸮⸮+⸮h⸮nV⸮ Y..	⸮⸮⸮j
⸮W⸮⸮⸮˹c⸮⸮S
⸮e⸮.⸮٠⸮⸮⸮⸮⸮Ҳ&⸮⸮C⸮⸮r⸮⸮xL⸮(*⸮]⸮T⸮⸮C⸮⸮SQU⸮⸮⸮t'(⸮*U⸮F⸮⸮*⸮%=O⸮⸮Ȧ+͡ ⸮ّ⸮e⸮.	⸮°j
⸮W}⸮a⸮⸮i⸮3⸮L*⸮e⸮.R⸮⸮ ⸮ҰL⸮⸮:&⸮⸮C⸮⸮rs⸮⸮⸮⸮0⸮	E5⸮D⸮%⸮
⸮RQj⸮⸮⸮⸮⸮:⸮⸮b⸮i%⸮FP*⸮e1⸮H%⸮O⸮JH⸮+⸮h⸮nV⸮A⸮.1 ⸮⸮⸮W⸮m⸮Z⸮A3S
⸮⸮w⸮⸮]⸮ ⸮⸮⸮⸮2&gӓH⸮
r.⸮¸L⸮⸮*⸮]T⸮*⸮}⸮E⸮U⸮I⸮⸮t'⸮ŢBZA⸮_T⸮T⸮⸮*M⸮_⸮ꊕ*⸮⸮+͡ ⸮Y⸮⸮er.	⸮⸮°C⸮⸮⸮_⸮⸮˹c⸮⸮Sj
e⸮.⸮n⸮⸮⸮⸮ ⸮⸮2⸮gӳH⸮⸮⸮⸮'⸮⸮0 IE5]D⸮%⸮
⸮⸮Q⸮⸮⸮b⸮⸮t⸮⸮(⸮*⸮}⸮LP
⸮%==T⸮C!⸮+⸮⸮ ⸮V⸮e⸮.	⸮°j
⸮W⸮mX⸮⸮⸮3⸮⸮⸮e⸮.Rչ ⸮ҰL⸮º⸮⸮5⸮H⸮
r⸮⸮¸L⸮⸮*⸮]⸮T⸮⸮C⸮j⸮UQ⸮,⸮⸮⸮:⸮EbB⸮A%_⸮P*⸮e1⸮S⸮%=⸮T⸮C!+⸮h⸮.V⸮Ae.1 L⸮!W⸮⸮X:⸮⸮37⸮j
eWJ⸮⸮⸮⸮⸮1⸮⸮º⸮⸮5SH⸮⸮⸮⸮:⸮E0⸮I⸮
⸮D⸮%⸮
_⸮Q⸮⸮⸮b⸮⸮t⸮⸮⸮ ⸮MP⸮e⸮⸮*⸮%=O⸮JHH⸮a.ҕ⸮⸮*⸮r	⸮°j
e⸮}m⸮Z⸮)A3S*⸮⸮WAJ]⸮⸮⸮ ⸮⸮2⸮'ӓH⸮
r⸮⸮⸮⸮L⸮(*⸮]T⸮*⸮}⸮E⸮QU⸮I⸮⸮t'⸮(⸮*⸮}⸮L⸮*⸮%⸮OU⸮Ȧ+⸮h⸮⸮a⸮*⸮⸮q L⸮0⸮!W⸮mXZ⸮⸮⸮⸮
⸮e⸮.Rչ ⸮⸮0L°'⸮:⸮⸮H⸮
```
When I attach the pin 18 and 23 the output is from then just crap!

Does this means that with this ESP the SPI Interface is not usable?

jdede, 2019-01-16:
That is weird... Can you access the Web UI?
Can you please post the versions of the libraries you are uing? I will try to reproduce this problem on my hardware.

PaulG4H, 2019-01-17:
Hello jdede,

here are the link to an 6MB [Archive](https://cloud.lxccu.com/index.php/s/Lw65EbZ3EPktp5p) of the Sketch and Librarys I currently use.

I do use this additional boards:
```
https://dl.espressif.com/dl/package_esp32_dev_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json
```

No I can't access the webUI...

PaulG4H, 2019-01-17:
Now I also Ordered an other [ESP32](https://www.dx.com/p/geekworm-esp32-worver-development-board-with-4mb-psram-wifi-bluetooth-low-power-consumption-dual-cores-esp32-2053524#.XEC55sZCcUE), not the SparkFun it's not the cost why 
I just like an external WiFi Antenna because the circuit printed ones are really poor.

It takes up to one month to get it but I am not in hurry...

jdede, 2019-01-17:
I updated all my libraries and tried to set this project up from scratch. Result: Currently, there seem to be several problems with AsyncTCP and the ESP32 core: DoorsignEPD (and other projects relying on these libraries) compile with several warnings (but no errors). The resulting program seems not to run properly. Unfortunately, AsyncTCP is neither using tags nor branches for the versioning which makes going back to another specific release quite hard.

I suggest to wait for some days and see, if the developers solve the problems.

PaulG4H, 2019-01-18:
Hello jdede,

I can compile the code with the libraries attached above, just the ArduinoJson need to be revertet to the latest Version 5 what is not Beta!

```
Build options changed, rebuilding all
Sketch uses 1098686 bytes (83%) of program storage space. Maximum is 1310720 bytes.
Global variables use 108888 bytes (33%) of dynamic memory, leaving 218792 bytes for local variables. Maximum is 327680 bytes.
```

jdede, 2019-01-20:
Hm, that is strange...
I updated the readme in PR #32 and stated the library versions of the major libraries. I also completely reset my system and started from scratch: It seems to work without any problems.

There seem to be some differences in the library versions between yours (from the zip archive) and mine. Can you please try updating Basecamp to the master of the git repository?

PaulG4H, 2019-01-30:
Fixed it...

Server Address do not start with http:// just enter the <ip or Hostname>  and check if the Firewall Rules are set correctly if you use different sub nets!

Today I got the new ESP32 [http://www.raspberrypiwiki.com/index.php/ESP32_WROVER-i_DevKit] which is working in the Arduino IDE after select ESP32 Wrover Board and set Baudrate to 115200 and switch the programmer to USBasp 

Thanks anyhow for the outstanding support!

jdede, 2019-01-30:
Good to hear that! The `http://`-issue is a nasty one! :+1: 
Have fun with your display!

closed 2019-01-30 by jdede