jamct/DoorsignEPD #36: E-Paper ESP32 Driver Board

borgelite, 2019-06-21:
Hallo, 
wer hat schon das E-Paper ESP32 Driver Board von Waveshare zum laufen gebracht?

qwykx, 2019-06-21:
Hallo @borgelite 

Ich nutze zwar nicht das genannte Board aber trotzdem:

Wie kann man dir behilflich sein?

borgelite, 2019-06-23:
Das Board hat ja eine andere Pinbelegung als das normale esp32 lolin Board. Ich finde im Code nur nicht die Einstellungen für die SPI Ansteuerung.

Laut dem manual https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board

wären das diese Pins:

SCK 13
DIN 14
CS 15
BUSY 25
RST 26
DC 27

Das Ding ist das, dass Board einen direkten Flachbandkabel Anschluss besitzt, 
und das ich nicht einfach die Pins umstecken kann.
Also muss ich das im Code irgendwo eintragen können.

qwykx, 2019-06-23:
Schau dir hier mal ab Zeile 85 das Beispiel für das ESP32 an :)
https://github.com/ZinggJM/GxEPD/blob/master/examples/GxEPD_Example/GxEPD_Example.ino

Ich denke damit kannst du was anfangen oder :)?

borgelite, 2019-06-23:
Das habe ich leider auch schon versucht . Ohne Erfolg ;(

Es funktioniert nur mit dem mitgelieferten Beispielcode.
https://www.waveshare.com/wiki/File:E-Paper_ESP32_Driver_Board_Code.7z

hardlog, 2019-06-26:
Hi,

hab das selbe Problem. Hab das ESP32-Driver Board von waveshape. Mit der Testsoftware von waveshape läuft es. Nur wie ich ggf. die doorsignEPD.ino ändern muss weiß ich nicht. Ich hab noch ein ESP32 von AZDelivery, da finde ich keine vernünftige Pin-Beschreibung (wo sind die D's ???) ...
Ist von euch schon jemand weiter?

hateblast, 2019-09-06:
Ich hatte dasselbe Problem mit dem ESP32 Driver Board und es folgendermaßen gelöst:

1. Pinbelegung im Code anpassen

GxIO_Class io(SPI, CHIP_SELECT, 27, 26);
GxEPD_Class display(io, 26, 25);

#define CHIP_SELECT 15

2. In der Setuproutine folgendes nach display.init() einfügen:

  SPI.end(); // release standard SPI pins, e.g. SCK(18), MISO(19), MOSI(23), SS(5)
  //SPI: void begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);
  SPI.begin(13, 12, 14, 15); // map and init SPI pins SCK(13), MISO(12), MOSI(14), SS(15)
 
Ich hoffe, dass das weiterhilft.
Viele Grüße

borgelite, 2019-09-10:
Vielen dank @hateblast  aber mein Display flackert leider nur bei der Aktualisierung.
Und über wlan kann ich es auch nicht erreichen.

jdede, 2019-09-10:
Ich denke, dass im Beispiel von @hateblast der SPI nicht sauber initialisiert wird bzw. die SPIs gemischt werden und es dann auch Intern Probleme mit dem WiFi gibt. Habe das nun nicht getestet, sollte aber mit zwei SPIs laufen. Hierzu gibt es einen Eintrag mit einer Lösung: https://github.com/espressif/arduino-esp32/issues/1219

Kurzfassung:
-  `#include "SPI.h"` hinzufügen
-  `SPIClass displaySpi(HSPI);`
- `GxIO_Class io(SPI, CHIP_SELECT, 17, 16);` in `GxIO_Class io(displaySPI, CHIP_SELECT, 17, 16);` ändern
-  `displaySPI.begin(13, 12, 14, 15);` im Setup hinzufügen

Wie gesagt: Nicht getestet, nur aus dem Kopf geschrieben.
Vielleicht klappt es ja.
Viele Grüße

j-ed, 2019-12-19:
Hallo zusammen, als vorweihnachtliches Bastelprojekt habe ich mir ebenfalls ein *e-Paper ESP32 Driver Board* von Waveshare als Basis für dieses Projekt angeschafft. Nach einiger Recherche konnte ich dank des von @hateblast beschriebenen Code-Schnipsels die ersten Hürden meistern und mein 4.2" e-Paper-Display ansteuern. D.h. die erwarteten Startmeldungen werden auf dem Display fehlerfrei angezeigt und es flackert auch nichts.
```
...
Not going to deep sleep. Reason: not in production mode
In setup mode
...
```
Mir ist es jedoch nach dem Verbinden mit dem WLAN "ESP32_...", eine IP-Adresse wird korrekt zugewiesen, nicht möglich auf die Weboberfläche zuzugreifen. Mit dem von @jdede beschriebenen Code-Änderungen bin ich leider nicht weiter gekommen, da dann keinerlei Meldungen mehr auf dem Display angezeigt werden.

Hat zwischenzeitlich jemand von Euch das Driver-Board ans Laufen bekommen oder noch eine Idee wie man die Fehlerursache identifizieren kann ? Ich bin für jeden Tipp dankbar, denn meine anderweitigen Recherchen sind bis dato im Sande verlaufen.

jdede, 2019-12-28:
Moin @j-ed ,
wenn die Status-Meldung angezeigt wird (ich denke mal du meinst `Connect to hotspot 'ESP32' with the secret...`), scheint die Ansteuerung des Displays zu funktionieren. Versuche mal bitte folgendes:

- Wenn du mit dem WLAN des ESP32 verbunden bist: Kannst du die IP des ESP pingen, also `ping 192.168.4.1`?
- Welchen Browser verwendest du? Bei Chrome scheint es Probleme zu geben. Versuche ggf. die Seite explizit mit einem `http` aufzurufen, also `http://192.168.4.1`. Firefox funktioniert bei mir am zuverlässigsten.
- Was ist der genaue Fehler beim Zugriff auf die Weboberfläche?

Vielleicht bringt es ja was.
Viele Grüße

j-ed, 2019-12-31:
Hi @jdede,
Deine Rückmeldung habe ich zum Anlass genommen mich erneut mit dem Driver Board auseinander zu setzen. Nachdem meine ersten Versuche der Konfiguration mit einem Galaxy S7-Telefon fehl geschlagen sind, habe ich testweise einmal ein Galaxy S9-Telefon ausprobiert und war überrascht, dass ich den gewünschten Konfigurationsdialog direkt angezeigt bekam.

Frei nach der Devise "Ein Reboot tut gut", habe ich dann mein Telefon neugestartet und siehe da, es wurde dem Gerät nicht nur eine IP-Adresse zugewiesen, sondern es war plötzlich auch eine Konfguration möglich - wer weiß wo es vorher geklemmt hat 😄 

Den gestrigen Tag habe ich also damit verbracht, mich mit dem Aufbau der WLAN-Verbindung und dem Ausprobieren der verschiedenen Beispielanzeigen auseinander zu setzen.

Zusammen gefasst laufen das **e-Paper ESP32 Driver Board** und das **4.2 inch e-Paper Display**  von Waveshare nun mit der von @hateblast vorgeschlagenen [Lösung](https://github.com/jamct/DoorsignEPD/issues/36#issuecomment-528901997) bei mir. Die von dir vorgeschlagene Anpassung führte bei mir bis dato leider dazu, dass die Anzeige nicht mehr aktualisiert wurde.

Vielen Dank vorerst für die angebotene Unterstützung!

HenrikSte, 2020-05-25:
> GxIO_Class io(SPI, CHIP_SELECT, 27, 26);
> GxEPD_Class display(io, 26, 25);
> 
> #define CHIP_SELECT 15
> 
>     1. In der Setuproutine folgendes **nach display.init() einfügen**:
> 
> 
> SPI.end(); // release standard SPI pins, e.g. SCK(18), MISO(19), MOSI(23), SS(5)
> //SPI: void begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);
> SPI.begin(13, 12, 14, 15); // map and init SPI pins SCK(13), MISO(12), MOSI(14), SS(15)

Danke dir, Made my day!