jamct/DoorsignEPD #37: Watchdog (AsyncTCP)

maltejahn, 2020-11-15:
Hi,

i try to use a e paper 4.2bwr (tried it with and without "bwr") with an AZ-Delivery ESp32 Dev Board. But, I always get the same error:
```
`Trying to connect ...
[E][AsyncTCP.cpp:722] connect(): error: -16
Not going to deep sleep. Reason: Not in production mode
Ongoing connection
In setup mode
Not going to deep sleep. Reason: Not in production mode
Ongoing connection
In setup mode
E (6042) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6042) task_wdt:  - async_tcp (CPU 0/1)
E (6042) task_wdt: Tasks currently running:
E (6042) task_wdt: CPU 0: IDLE0
E (6042) task_wdt: CPU 1: IDLE1
E (6042) task_wdt: Aborting.
abort() was called at PC 0x400ecf77 on core 0

Backtrace: 0x4008c970:0x3ffbe170 0x4008cba1:0x3ffbe190 0x400ecf77:0x3ffbe1b0 0x40084c49:0x3ffbe1d0 0x40185107:0x3ffbc8f0 0x400ee8fa:0x3ffbc910 0x4008a839:0x3ffbc930 0x40089055:0x3ffbc950

Rebooting...
ets Jun  8 2016 00:22:57
```
`

Backtrace
`0x4008c970: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008cba1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400ecf77: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x40185107: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400ee8fa: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008a839: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40089055: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

I tried to follow the guide, but I`m a little lost
GxEPD 3.1.0
Basecamp 1.8.0
Adafruit_GFX 1.10.2
Asynctcp - latest via git clone
ArduinoJSON 5.13.5
ESPAsyncWebServer - latest via git clone

If i disable 
`display.update(); ` 
at 
`onDisconnectHandler`
it seems no to crash.

How can I debug this? Does the display take too long? Is it a bad Wifi connection,...

GxEPD Example with 4.2 and 4.2bwr Display does work - so no hardware issue?

jdede, 2020-12-02:
Hi Malte,
hm, sounds weird. Never had watchdog issues with this code. I guess that this is the result of

- Outdated libraries: Basecamp for example is outdated since at least two years
- Newer versions of the ESP toolchain
- Tons of old code

I guess the whole project should be rewritten using more recent libraries. I haven't worked with Arduino for quite a while and this issue doesn't look like a no-brainer.

maltejahn, 2020-12-03:
Hi,

i did it (not) the professional way
I edited the 
CONFIG_ASYNC_TCP_USE_WDT = 0
in AsyncTCP.h to disable the WDT... 

Till now, i havent had any issues and it works. Maybe the problem comes with the display i use (4.2", 3 color) which maybe need a long time to refresh. Maybe its the network connection - which i doubt because the WLAN AP is in the same.

Probalby i will "downgrade" the Espressif Arduino libraries..

jdede, 2020-12-03:
Hm, okay. Hope that this works. I have heard from others that in this case the code just freezes... Good to hear that it seems to work for you.

Best regards
Jens