jamct/DoorsignEPD #41: Seit PHP8 funktioniert der Server nicht mehr

dpuschek, 2024-09-27:
Hallo,
vor einem Jahr habe ich Debian nach Bookworm upgegraded. PHP hat sich dabei von Version 7.4 -> 8.2 aktualisiert. Seitdem funktioniert das Display nicht mehr.
Ich habe das Problem im Forum auf php.de geschildert, wo mir aber kaum geholfen wurde. Bin ich wirklich der letzte, der das EPaper-Display noch nutzt?

https://www.php.de/forum/webentwicklung/php-einsteiger/1611140-php-script-l%C3%A4uft-nach-systemupdate-nicht-mehr-uncaught-typeerror-imagedestroy

smmmo, 2024-09-27:
Hi,

it's working for me with php 8.2.

Here's the output of the changes that I made (in 2022) to make it work again:
[git_diff.txt](https://github.com/user-attachments/files/17168995/git_diff.txt)

Hope that helps

dpuschek, 2024-09-28:
thank you. I already did that changes, but it didn't solve the problem at all. I always get a black display with it. My black and white PNG file could no longer be converted correctly :(

jdede, 2024-09-28:
Hey, cool: Someone is still using this code. Haven't touched it for quite a while but tried to fix the main PHP-issues (and I know again why I hate this language...) :-D

@dpuschek can you check the PR #42 and report if this works for you?

dpuschek, 2024-09-29:
Thank you. Unfortunately I still got black images at the display, if debug=false. I attached both GET-requests and named it .png (otherwise I cannot attach binary files here)

[image]
[image]

Here are my apache logs for one request:
access.log
192.168.1.112 - - [29/Sep/2024:08:50:51 +0200] "GET /server/index.php?debug=false&display=7.5&content=static_image&display=7.5 HTTP/1.1" 200 30938 "-" "-"

error.log
[Sun Sep 29 08:50:51.018955 2024] [php:warn] [pid 78161:tid 78161] [client 192.168.1.112:52455] PHP Warning:  Undefined array key "scale" in /home/wwwdir/server/index.php on line 63
[Sun Sep 29 08:50:51.073271 2024] [php:warn] [pid 78161:tid 78161] [client 192.168.1.112:52455] PHP Warning:  Undefined array key "red" in /home/wwwdir/server/index.php on line 131

jdede, 2024-09-29:
Okay, just pushed another update to the PR. Can you please check again?

Weird: You are just having warnings in your `error.log`. There might be a problem in the binary representation for EPD or, more precisely, in the function `rawImage()`. To debug that, I have to first get one of the displays and an ESP in my fingers...

dpuschek, 2024-09-29:
error.log warnings are gone! But still black display. Yes, the rawImage() function doen't work as it did in PHP7. I don't understand much about PHP, but there must changes at GDImage class at PHP8:
https://phpbackend.com/blog/post/php-8-0-gdimage-object-is-used-instead-of-resource-type

Unfortunately I cannot fix it myself and PHP community don't want to help me, as I already mentioned in my first posting here.

"GDImage class object instead of 'resource' type." I have no idea about OOP Programing. It is too complicated for me :(

jdede, 2024-09-29:
Ah, good catch! Thanks! Updated the PR accordingly. There seems to be also an issue with the background color. Hope that the new version works. Can you please check it again? Afterwards, I can merge it into the main module.

dpuschek, 2024-09-29:
New version work! But I'm only using 'static_image'. The other "contents" stuff I never used.

jdede, 2024-09-29:
Okay, I briefly checked most of them. Should work. Lets see if someone complains after the merge.

Thanks for your support!

closed 2024-09-29 by jdede

smmmo, 2024-09-29:
> thank you. I already did that changes, but it didn't solve the problem at all.
So sorry, my mistake. My workaround in 2022 was to run the script inside a docker container with php8.1 and mount the directory of DoorsignEPD to use it inside the docker container. For this, the 2 fixes were enough.

However, thanks to @jdede for the proper fix!

commit cb8c8ebd (2024-09-29): Merge pull request #42 from jdede/fix-php8

Adapt code for newer PHP versions. Fixes #41